adafruit/tinyuf2 #379: Download missing for lolin s3 mini?

uecoding, 2024-03-18:
I'm trying to download the bootloader for lolin s3 mini from [downloads](https://circuitpython.org/board/lolin_s3_mini/) it shows this link but it seems like the file doesn't exist anymore:
https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-lolin_s3_mini-0.18.1.zip

Where can I find the correct bootloader?

dhalbert, 2024-03-18:
It looks like there never was a Lolin S3 mini version of the bootloader created. There is a regular "Lolin S3" but not mini.

You can load the .bin: you don't have to use the UF2. The regular S3 version of the bootloader _might_ work, but I haven't looked to see what the board differences are.

uecoding, 2024-03-18:
Well I erased the flash and uploaded combined.bin of regular lolin s3 and then did the reset but it doesn't show up as a new drive.

uecoding, 2024-03-19:
Oh okay. I used the online installer and selected "Upgrade CircuitPython 8.2.10 Bin Only". Now it works

closed 2024-05-24 by brentru