adafruit/tinyuf2 #395: STM32F4 board: stm32f411ce_blackpill wont compile

Rulaman, 2024-07-13:
### Operating System

Linux

### INFO_UF2.TXT

-none-

### What happened ?

Cannot compile the board as in the tilte described

### How to reproduce ?

.../tinyuf2$ python3 tools/get_deps.py stm32f4
.../tinyuf2$ python3 tools/get_deps.py --board stm32f411ce_blackpill
.../tinyuf2/ports/stm32f4$ make BOARD=stm32f411ce_blackpill all

### Debug Log

AS startup_stm32f411xe.o
CC ghostfat.o
CC images.o
CC main.o
.../tinyuf2/src/main.c:33:10: fatal error: tusb.h: No such file or directory
   33 | #include "tusb.h"
      |          ^~~~~~~~
compilation terminated.
make: *** [../rules.mk:92: _build/stm32f411ce_blackpill/obj/src/main.o] Error 1

Rulaman, 2024-07-18:
In the compile description there is simply a line missing.

`git submodule update --init --recursive`

Execute this command from the directory, where the repo has been cloned.

hathach, 2024-07-18:
no, you need to run, it is mentioned in the readme 
```
python tools/get_deps.py stm32f4
```

Rulaman, 2024-07-18:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALXED7UABL62TC54UM73QQ3ZM6DHFA5CNFSM6AAAAABK2IK75WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUFI5VCA.gif]

Hi Ha,

i did run the command

> *python tools/**get_deps.py** stm32f4*

but it did not fetch the necessary repositories like tinyusb and uf2.

It was not clear enough, that I have to clone the repo with submodules, which I didn`t.

That is the point that lacks in the docu.

Greetings
Rulaman

hathach, 2024-07-18:
ah you are right, let me update the doc