adafruit/tinyuf2 #401: Lolin S3 mini has wrong link for tinyuf2

marcobrianza, 2024-08-06:
### CircuitPython version

```python
CircuitPython 9.1.1
```

### Code/REPL

```python
we don't get to run the code
```

### Behavior

nothing

### Description

in this board page https://circuitpython.org/board/lolin_s3_mini/
the link for the tinyuf2 is this:
https://github.com/adafruit/tinyuf2/releases/download/0.18.2/tinyuf2-lolin_s3_mini-0.18.2.zip

the file does not exist and the web installer locks

dhalbert, 2024-08-06:
transferred issue to tinyuf2 repo. tinyuf2 needs a new release to make new boards available.

@hathach is it fine just to do a 0.18.3 release, or is there any reason to hold off on that? Is there anything to do manually. We can do it if you are busy.

When this is done, the release version needs to be bumped in https://github.com/adafruit/[circuitpython-org](https://github.com/adafruit/circuitpython-org)

marcobrianza, 2024-08-07:
[image]
it may just be that the file is link need to be renamed 
https://github.com/adafruit/tinyuf2/releases/download/0.18.2/tinyuf2-lolin_s3-0.18.2.zip

marcobrianza, 2024-08-07:
BTW also this board has the same issue 
https://circuitpython.org/board/lolin_s3_pro/
where the file link is not found
https://github.com/adafruit/tinyuf2/releases/download/0.18.2/tinyuf2-lolin_s3_pro-0.18.2.zip

marcobrianza, 2024-08-07:
going further with the reaserch I found that "Lolin S3" "lolin S3 Pro" and "Lolin S3 mini" may be 3 different designs and need different tinyuf2 after all. I have checked the PID VID and found that S3 pro and mini have the same id

[image]

[image]

hathach, 2024-08-08:
@dhalbert just make an release for 0.19.0, but ci seems to have some issue. I will check and fix it asap.

hathach, 2024-08-09:
fixed ci and re-release 0.19.0

marcobrianza, 2024-08-11:
I can aslo test loin S3 pro but there is no new release for it

tannewt, 2024-09-05:
I fixed the CI issue with release 0.20.1 and https://github.com/adafruit/circuitpython-org/pull/1469 updates the website to use it.

closed 2024-09-05 by tannewt