adafruit/tinyuf2 #408: Release failed to upload esp32 binaries to aws

hathach, 2024-09-04:
### Operating System

Others

### INFO_UF2.TXT

n/a

### What happened ?

when making a new release, build failed when trying to upload artifacts/binaries to aws with esp32 build.

https://github.com/adafruit/tinyuf2/actions/runs/10703503616/job/29674841712

I am not familiar with this @makermelissa since you fixed this a couple of time previously, could you please lease a hand whenever you have time. Thank you. 

### How to reproduce ?

make a new release

makermelissa, 2024-09-04:
I should be able to look at this today.

makermelissa, 2024-09-04:
I see "Unable to locate credentials".

tannewt, 2024-09-05:
This is fixed by 23362cd. Secrets need to be explicitly passed to sub-jobs.

closed 2024-09-05 by tannewt

hathach, 2024-09-05:
thanks @tannewt for the fix

makermelissa, 2024-09-06:
Thanks @tannewt. I wasn't aware of this new option.