adafruit/tinyuf2 #427: Flash clock frequency changed for imxrt10xx boards.

robert-hh, 2025-01-27:
### Operating System

Linux

### INFO_UF2.TXT

TinyUF2 Bootloader 0.20.1-25-g46de518-dirty - tinyusb (0.16.0-1090-g3eea46056)
Model: NXP RT1020 EVK
Board-ID: MIMXRT1020-EVK-revA
Date: Jan 27 2025
Flash Size: 0x00800000 bytes

### What happened ?

The bootloader changes the flash clock frequency for the imxrt10xx boards. For most boards it is lowered from 100Mhz to 60Mhz or 30MHz. For the Teensy it it raised from 100Mhz to 133Mhz. Is is possible either

- not to lower the frequency? I made trial changes for 100Mhz and it still works.
- set it only when flash writes are happen and not on every boot?
- set it back?

The INFO.UF2.TXT is just a sample from one board. The issue applies to all imxrt10xx boards.

### How to reproduce ?

Compare the flash clock frequencies before and after the flash configuration was changed in the bootloader.

robert-hh, 2025-01-27:
Looking at the code of main.c it seems that the flash mode is re-configured only if the board is in  DFU mode & the firmware image is to be uploaded. That would be option 2 about. But flash is slow even if the app is called.
Note: how do I get access to the LOG messages?

hathach, 2025-02-06:
thanks, I think this is already fixed by https://github.com/adafruit/tinyuf2/pull/428. If so we could close this one ?

robert-hh, 2025-02-06:
Yes, it was merged by Scott. Can be closed.

closed 2025-02-06 by robert-hh