adafruit/tinyuf2 #432: CircuitPython 10  will change partition table for 4MB ESP32-S3 flash boards

dhalbert, 2025-02-07:
Hi @hathach -- Our plan for CircuitPython 10 is to change the partition table for ESP32-S3 4MB flash boards from `partitions-4MB.csv` to `partitions-4MB-noota.csv`. Currently there is not enough room in the `ota_0` partition to include BLE support.

These are the boards that would be affected:
```
`
adafruit_feather_esp32s3_reverse_tft
adafruit_feather_esp32s3_tft
adafruit_qtpy_esp32s3_n4r2
deneyap_kart_1a_v2
lolin_s3_mini
magiclick_s3_n4r2
waveshare_esp32_s3_zero
```
We will warn CircuitPython users in advance that they will need to upgrade their bootloaders. We are wondering for these boards:
- should tinyuf2 build two versions of bootloaders for each of these boards, with and without `ota1`, and produce two sets of artifacts?
- or should there be new board definitions, e.g. `adafruit_feather_esp32s3_no_ota`
- or whether we should make a major version change (1.0.0 and beyond) that makes this incompatible change.

Any other ideas? Your comments are appreciated.

There may be non-CircuitPython users of tinyuf2 who would be affected by such an incompatible change. If you are such a user, could you speak up?

tannewt, 2025-02-07:
Thanks for writing this up Dan.

I want to point out that we'd like to do this before CircuitPython 10. It'll work with CP9 too. It is needed for 10 so that we can have a larger firmware image. When the bootloader hasn't been updated, then the bootloader should error.

tannewt, 2025-03-27:
@hathach Would you mind doing this now? We'd like to start release 10 pre-releases.

tannewt, 2025-03-31:
We need to ensure that the Arduino options still align with this change.

hathach, 2025-04-03:
sure, I will also update the Arduino for our board to have signle OTA as default.

hathach, 2025-04-08:
implemented by #442 , since the partition table is dictated by partition-binary at 0x8000. Currently we don't erase/rewrite patition table (a bit dangerous), so we cannot switch from no-ota to ota with just using drag-and-drop update-tinyuf2.uf2. It needs to be done using esptool.py with combined.bin. The actual binary for tinyuf2 partition does not channge at all.

dhalbert, 2025-04-17:
@hathach We decided that we want to remove the second OTA partition now for **all** 4MB boards, both S2 and S3. Should I make a new issue for that?

We are also going to do that for the non-UF2 boards, like ESP32, C3, etc.

Could you make that release `1.0.0`? Maybe let's drop the 0.30.0 release, since it is partial. I was thinking of renumbering that release anyway. Thanks.

hathach, 2025-04-18:
@dhalbert there is no need to create a new issue, let me check all the s2 and move it to the non-ota partition.

hathach, 2025-05-15:
implemented, in release 0.31.0 for board with 4MB flash (s2 and s3), there will be combined.bin and combined_ota.bin.

[image]

dhalbert, 2025-05-15:
> implemented, in release 0.31.0 for board with 4MB flash (s2 and s3), there will be combined.bin and combined_ota.bin.

So `combined_ota.bin `includes `ota_0` and `ota_1`, and `combined.bin` includes only a twice-as-large `ota_0`? That looks like what is happening.

I noticed the file list above only has one `partititon-table.bin`, but I would expect two, one with `ota_0` and `ota_1`, and one with only `ota_0`. Which partition layout is in `partition-table.bin`?

hathach, 2025-05-15:
yeah, the default is single bank (noota). I can add the partion-table-ota.bin as well. in the follow up PR.

hathach, 2025-05-16:
released as 0.32.0