adafruit/tinyuf2 #451: STM32L4 Port Shows GitHub Automation Build Errors

BionicCode, 2025-06-26:
### Operating System

Windows 11

### INFO_UF2.TXT

-

### What happened ?

The repository under [ports/stm32l4 ](https://github.com/adafruit/tinyuf2/tree/master/ports/stm32l4) shows build errors from GitHub automations:

[image]

Can you please validate the STM32L4 port to check whether these errors are relevant or need to be fixed?

Thank you very much! I want to use your bootloader but I need to ensure that it operates properly. It's difficult for me to understand the relevance of these GitHub errors. Last check has been a while so there are no logs available anymore.

### How to reproduce ?

Navigate to [ports/stm32l4 ](https://github.com/adafruit/tinyuf2/tree/master/ports/stm32l4) and see run status:

[image]

### Screenshots

[image]
[image]

dhalbert, 2025-06-30:
This broke in the 0.33.0 release as well.

hathach, 2025-07-01:
> This broke in the 0.33.0 release as well.

it build just fine, ci failed due to file name/path change. I will check.

hathach, 2025-07-02:
too obsolete, current version build just fine

closed 2025-07-02 by hathach

BionicCode, 2025-07-02:
What is it that makes the CI build fail for this port and not for others?

hathach, 2025-07-03:
6 months old is too obsolete, current code build L4 wihtout issue on CI

hathach, 2025-07-03:
https://github.com/adafruit/tinyuf2/actions/runs/16029529864/job/45226219329

BionicCode, 2025-07-03:
I have just forked your repo and then enabled the original actions. Could you do me a favor and modify the [ports/stm32l4/README.md](https://github.com/adafruit/tinyuf2/blob/master/ports/stm32l4/README.md) like addinig a whitespace etc. and then commit to trigger the CI build of this particular port? This is what fails.
Also, if it doesn't fail it will remove the error batch from the port which is also nice. Thank you

BionicCode, 2025-07-03:
> Please try to get familliar with github action status and how github web display status for the last commit in a folder view.

Yeah, I think I understand it. In my original post I asked _"Can you please validate the STM32L4 port to check whether these errors are relevant or need to be fixed?"_. The whole idea of contacting you was to ensure that these errors are not relevant. I don't have time to waste like configuring a custom board just to find out ((probably late in the development process) that the bootloader is kind of crooked. I did not know what exact actions where failing, except that I was able to reproduce those exact errors using thhe lates 'master' version. 

I've posted the messages to give you the chance to glance over it and confirm that these errors are not relevant. I'm just not used to the idea that such errors are not relevant in general and are left unfixed. So I asked you. I know that the port itself builds.

Thanks for your time!