adafruit/tinyuf2 #461: combined.bin missing for M5Stack Dial

ishotjr, 2025-07-18:
### CircuitPython version and board name

```python
n/a; M5Stack Dial
```

### Code/REPL

```python
n/a
```

### Behavior

file note found at https://adafruit-circuit-python.s3.amazonaws.com/bootloaders/esp32/m5stack_dial/tinyuf2-m5stack_dial-0.35.0-combined.bin

```xml
<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>bootloaders/esp32/m5stack_dial/tinyuf2-m5stack_dial-0.35.0-combined.bin</Key>
<RequestId>72H3P2CEBHJY9JF6</RequestId>
<HostId>x0dDo+1+LIr0f8PEepejXh+yu2500mFsxVpT+5+L5mFoUQC5zzpY2ohUj+omVr/Q8PokZIQHUQU=</HostId>
</Error>
```

### Description

Navigating to the link provided by `DOWNLOAD BOOTLOADER combined.bin` on https://circuitpython.org/board/m5stack_dial/ results in the above `NoSuchKey` error

dhalbert, 2025-07-18:
There is no board definition for `m5stack_dial` in the tinyuf2 repo. Transferring this issue to there, though I think we might await a community contributor of a board def. There may be another one that would work.

hathach, 2025-12-18:
PR is welcome